BruceDevices/firmware #1092: M5StickC Plus2 doesnt turn on while nrf24 is connected.

Drobeled, 2025-04-15:
The title says it all. also, i tried mirrorint the pins, tried to use nrf while stick is turned on but instead it says: “nrf24 not found”, restarting doesnt help.

RossFallESP32, 2025-05-20:
You must turn on the device first before installing it, or you must connect the charger to stimulate and supply power to the nrf24 during startup because the c plus2 does not have enough voltage for the c plus2 and nrf24 to start working together.

Drobeled, 2025-05-20:
i tried it too, but the problem was the nrf itself. i fixed it by getting a new one.

Huzzla101, 2025-05-20:
Admit that you didnt get a new one but you had one wire connected wrong on
ur nrf or device..... =) HAHAHA That happened to me at least , then it
wont  turn on before you fix it ....