BruceDevices/firmware #1095: Make Smoochie Support To ESP32-WROOM-32

SteveThe00, 2025-04-16:
I'm recently buying an Esp32 wroom 32 and 1.9 inch tft and i don't know which one firmware that can flashed on esp32, so someone help me

SteveThe00, 2025-04-16:
@pr3y

tegtheevi, 2025-04-16:
You shouda have bought the cheap yellow display it saves you a lot of time & trouble

SteveThe00, 2025-04-17:
> You shouda have bought the cheap yellow display it saves you a lot of time & trouble

i was buying cheap yellow display but it broke because i was wrong when soldering

tegtheevi, 2025-04-17:
Whats you objective with the device 
I can help you

SteveThe00, 2025-04-17:
> Whats you objective with the device I can help you

My objective is just to install bruce on it an i will enjoy it on school because my school wont allow the student to play handphone

SteveThe00, 2025-04-17:
> > > Whats you objective with the device I can help you
> > 
> > 
> > My objective is just to install bruce on it an i will enjoy it on school because my school wont allow the student to play handphone
> > https://youtu.be/lKPhp69dbVE?si=QXfdBCeGxkK-13AI

Different board in that video he using esp32 s devkit n16r8 and i using esp32 wroom 32

tegtheevi, 2025-04-17:
> Different board in that video he using esp32 s devkit n16r8 and i using esp32 wroom 32
Get an esp32 s3 then and just take it easy

SteveThe00, 2025-04-17:
> Whats you objective with the device I can help you
>
>
> My objective is just to install bruce on it an i will enjoy it on school because my school wont allow the student to play handphone [...]

I dont have money to buy that and in indonesian the price look like going high

SteveThe00, 2025-04-18:
> Whats you objective with the device I can help you
>
>
> My objective is just to install bruce on it an i will enjoy it on school because my school wont allow the student to play handphone [...]
should i buy 2.4/2.8 inch ili9341 tft lcd?

tegtheevi, 2025-04-18:
> should i buy 2.4/2.8 inch ili9341 tft lcd?

Try to fix your CYD and get an sd card sniffer to avoid soldering directly on the board if you wanna use nrf24/cc1101