BruceDevices/firmware #1098: Add Targeted Handshake Capture

p0nyb0yt00st0ned, 2025-04-17:
Is your feature request related to a problem? Please describe.
No it is related to adding a feature

Describe the solution you'd like
Hello, i think it would be a great idea to add a access point handshake sniffer/capture. Brucegotci is great but when i want to target a specific network to try and get a handshake on it could take long with Brucegotchi. Thank you!

**Describe alternatives you've considered**
I've seen another esp32 wifi pentesting github repo that allows targeted handshake capture
[](https://github.com/risinek/esp32-wifi-penetration-tool)

**Additional context**
I think this would be a great addition to Bruce's suite of amazing tools. Please consider adding this feature it would be so beneficial.

bmorcelli, 2025-05-26:
You can do this:
* Wifi > Target Atks> [choose AP] > Information
* Take note of the channel
* Wifi > RawSniffer > change to the channel of the targetted AP
* wait, and you can turn on and off Deuth on this particular channel.

cop1lot1, 2025-05-29:
Why MAX_CHANNEL in sniffer is 11? What if target on 13 channel?