BruceDevices/firmware #1105: Openhaystack not working on T-Embed CC1101

stefexec, 2025-04-19:
Environment:

    bruce version: 1.9.1
    T-Embed CC1101

Steps to reproduce:

    Copy a file with the generated Openhaystack private key to the fs of the device
    Open Others -> Openhaystack

Expected behavior:

    Data for openhaystack was sent

Actual behavior:

    Data for openhaystack was not sent
    In logs we see some errors:
    gap register error: ESP_ERR_INVALID_STATE couldn't set random address: ESP_ERR_INVALID_STATE couldn't configure BLE adv: ESP_ERR_INVALID_STATE

stefexec, 2025-04-19:
This also happens on f612a0e

pr3y, 2025-04-21:
Should be good now after https://github.com/pr3y/Bruce/pull/1110
If problem persists on Beta build, create another issue pls

closed 2025-04-21 by pr3y