BruceDevices/firmware #1106: Can't pair BLE Media commands on T-Embed CC1101

IgnasFalcon, 2025-04-19:
Describe the bug:
Can't pair BLE for Media Commands

Steps To Reproduce:
Go to "Media Cmds" and pair
- Device used: T-Embed CC1101

Expected behavior:
When paired the device will know it and show the menu for commands

Actual behavior:
When paired the device doesn't acknowledge it and still waiting to be paired

Additional context/logs:
Paired with Windows 11 computer with a new T-Embed CC1101. Done in the current latest version. (The bug is still in the beta version too)

ben7147, 2025-04-23:
Same with M5 stick, and bad ble doesnt work neither

2xTheTap-dev, 2025-05-20:
I had a similar experience. However, I noticed that if you pair the bruce using the badble module or btkeyboard with another device (phone, pc) then when you go back into the bluetooth menu you will be able to use the media control option.

bmorcelli, 2025-05-26:
following on https://github.com/pr3y/Bruce/issues/1211