BruceDevices/firmware #1115: Bricked a brand new M5 Stick C Plus

HRPlacehold, 2025-04-22:
**Describe the bug:**
Burned to brand new M5 Stick C Plus and now the device is bricked

**Steps To Reproduce:**
No idea, burn it?
- Device used: M5 Stick C Plus

**Expected behavior:**
It would not be dead

**Actual behavior:**
It died

**Additional context / logs:**
Got nothing because it's dead and won't turn on

rennancockles, 2025-04-23:
What do you expect from us with this issue? To unburn your device? By the way, what do you mean by burn? How did you do it?

tommyfart, 2025-04-23:
charge it

bmorcelli, 2025-04-24:
Find yourself a male-male jumper wire.

* Connect on end to GND, the other end to G0, inthe stickCPlus header
* Plug the usb Cable
* Flash the correct firmware (if your stickCPlus is yellow, it is the StickCPlus2)
* Unplug the jumper Wire
* Unplug the USB cable
* Hold the power btn for 10s
* Plug the USB cable again, it must turn on

closed 2025-05-24 by bmorcelli