BruceDevices/firmware #1121: Feature Addition (Targeted Handshake Sniff,  Better Bad BLE Config?)

p0nyb0yt00st0ned, 2025-04-23:
**Is your feature request related to a problem? Please describe.**
No it is not related to a problem.

**Describe the solution you'd like**
There are a few Ideas I would love to see implemented into Bruce firmware.
1. Targeted AP handshake sniff (With the ability to turn deauth on or off during the sniffing).
2.  Better Bad BLE Configuration (I would love to be able to change my Bluetooth device name when i use bad BLE not just the standard "Keyboard_(Whatever)").

**Describe alternatives you've considered**
There are many different esp32 based firmwares that i have tried that have these features implemented. it would be amazing if this could be added in the future.

**Additional context**
MESSAGE TO THE COMMUNITY:
If you would like to see these changes happen please like this request and also make requests for these changes and any other implementations you would like to see

HenryTheAddict, 2025-04-24:
add skid mode that makes you a very cool hacker

bmorcelli, 2025-04-24:
> Targeted AP handshake sniff (With the ability to turn deauth on or off during the sniffing).

Follow these steps:
* Wifi Atks> Target Atks> (choose your target)> Information 
* Write the Channel number on a paper, or memorize it
* Get back to Main Menu
* Wifi> RawSniffer 
* Change the channel by pressing next or prev button, until you reach the channel you wrote down (or memorized)
* Press Ok to options, where you can turn Deauth Off, save only EAPOL/Handshashakes
* It will keep sending deauth every 30s, unless you turn it off.

closed 2025-04-24 by bmorcelli