BruceDevices/firmware #1124: Battery percent doesnt refresh automatically

ben7147, 2025-04-23:
**Describe the bug:**
Battery percent doesnt refresh automatically on main menu.
**Steps To Reproduce:**
Use latest build
- Device used: M5 Stickc Plus2

**Expected behavior:**
Battery percent and status refreshes automatically.
**Actual behavior:**
Battery percent and status doesnt refresh automatically, only after going in and out a menu.

**Additional context / logs:**
Not needed

HenryTheAddict, 2025-04-23:
same on t embed cc1101.

Gosheto1234, 2025-04-23:
the whole battery drawing is made that way , so its refreshed only when moving

ben7147, 2025-04-23:
It worked on older version. I remember there was a 30 sec delay or smth like that

bmorcelli, 2025-05-14:
it is refreshing every 30s now.. fixed on Beta in 14/05

closed 2025-05-14 by bmorcelli