BruceDevices/firmware #1135: CC1101 refuses to flash

Z4Vendetta, 2025-04-27:
CC1101 refuses to flash.

Following Bruce flash instructions: 

- Removed battery. 
- Began Bruce install, opened serial port selection page.
- Held down center boot button and then plugged in USB-C cable. No change. 
- Repeated above steps but this time also pressed reset button while holding boot button. No change. 
- Attempted different usb-c cable. Attempted different usb-c computer port.  No change.

CC1101 starts up OK normally without boot button or any other intervention. 
Does not respond to any attempt to flash.

Z4Vendetta, 2025-04-27:
SOLUTION: It seems that the issue behind the above dilemma was that I was not holding BOTH the reset button AND the boot button at the same time. 
BOTH buttons must be held down at the time the usb c cable is connected. I suppose I had considered it unfathomable with sausage fingers. Thankfully my 9 year old was happy to assist.

closed 2025-04-27 by Z4Vendetta

HenryTheAddict, 2025-04-30:
even dads are using this wow!