BruceDevices/firmware #1136: T-Embed CC1101 battery drainage

Ahmadkorhani, 2025-04-27:
I don't know if this is normal or not as I don't have another device to test. The battery can barely last around 20 Hours in sleep mode.
It is brand new T-Embed. I tried with official and beta release. Both have same problem.

First I thought there must be some hardware fault that make it consume battery while in sleep mode, but I have tested every single module and they are all working.

Do I have faulty battery or it is normal ?

Sor3nt, 2025-04-27:
I have the same issue here. 24h and the device is empty

dexxter00, 2025-04-27:
I can confirm. 
Device in **Deep Sleep** -> battery drainage in ~24 hours
Device in **Power Off** state -> battery on 100% after 7-8 days, but device will not come on with any of the keys, needs USB plugged in to power on

andreock, 2025-05-09:
DeepSleep still consumes some current to mantain RTC controller on and wakeup device on button press.
If you poweroff the T-Embed CC1101 you will need to press the PWR button(on the PCB) for about 3 seconds to wakeup battery controller.

bmorcelli, 2025-05-09:
The biggest problem is that tft, CC1101 and PN532 do not enter in deepsleep mode.. so they are consuming energy..

We need to figure out how to turn them off

andreock, 2025-05-09:
PN532 have hard and soft poweroff mode on I2C. I have the patch to implement it. Also cc1101 have some sort of shut down mode AFAIK

bmorcelli, 2025-05-10:
Yes, they have.. I saw it on the datasheets this week.. but didn't have time to mess with it

Don't forget to add the methods into the device classes in the CC1101 lib on my fork and PN532 on rennan's fork

andreock, 2025-05-10:
I opened a PR for PN532 library: https://github.com/rennancockles/PN532/pull/2

I will try to add powerDown also for CC1101 soon.