BruceDevices/firmware #1143: LilyGO T-Embeded CC1101 - Can't connect to WebUI (bruce.local)?

VVilku, 2025-04-29:
I can't get into the WebUI either. I have no idea anymore.
Bruce version 1.9.1
Hardware: liliyGo T-Embeded CC1101

After starting the WiFi AP, I connect to the created network (BruceNet) using my android phone. 
The packet connection is disabled.

In the browser I am not able to open the page: http://bruce.local.

I tried to check with nmap (termux) for what IPs are on the network 172.0.0.0/24.
I found two: 172.0.0.1 and 172.0.0.2. Neither of them have any ports open.

I've done several reboots and also pulled out the SD card. Nothing helps.

Any suggestions?

br/
VVilku

HalfMatt, 2025-04-29:
Switch to beta firmware

VVilku, 2025-04-30:
okeeey. Thanks

closed 2025-05-15 by bmorcelli