BruceDevices/firmware #1145: NRF24  Jammer Mode

kosmostrater, 2025-04-29:
**Describe the bug:**
When in the NRF24 Jammer 
If you Press Next or Prev , it changes the mode but It over writes  the previous mode on the display without  clearing it so it become a jumbles mess.

**Steps To Reproduce:**
NRF24, Jammer, Then press Next or Prev , the more times you presss the worse it gets.
Same with CH Jammer the Channels don't clear so it becomes 2 solid blocks and you cant tell what channel your on.

- Device used: 
esp32-2432s08  CYD
tried both Bruce 1.9.1 and Beta

**Expected behavior:**
Should clear the previous mode before displaying the new mode.

[image]

kosmostrater, 2025-04-30:
I have Bruce loaded on a T-display S3 (beta version) . Connected up the NRF24 and the display section was fine, so its just on the CYD. 
Not sure that helps, but I was curious :)

LittleHackid, 2025-05-01:
I get the same error, maybe it's only on the esp32 CYD

Huzzla101, 2025-05-01:
Man.... The last NRF24 update..... I still wake up at night believing i was
allowed into the secret bluetooth heaven and from there i just rock..... EL
MAGICO !!!!! Now we only need a NRF24 reader that can read the transmission
between the devices.. I try to find  Mr Vadim who made the one for flipper
and the world famous mousejacker  that can track a mouse dingle traffic
then find the device id and even the signals it send outs, then copy the
device id and  then send a "bad usb "with it like a .txt file etc..---

LittleHackid, 2025-05-01:
@kosmostrater this is the solution, you must put it in nrf_jammer.cpp
tft.setCursor(10,100);
uint16_t bgColor = ILI9341_BLACK;
tft.fillRect(10, 100, 250, 16, bgColor);

[image]

[image]

kosmostrater, 2025-05-01:
Sure ; that is  what I would expect (i had the same issue on projects) 

After I make the change, how I recompile the code and upload? 

Also add here as well  to clear the CH#

       while (!check(SelPress)) {
            if (redraw) {
                tft.setCursor(10, 35);
                tft.setTextSize(FM);
                tft.println("NRF X Jammer ");
                tft.setCursor(10, tft.getCursorY() + 25);
                tft.println("STATUS : ACTIVE");
                tft.setCursor(10, 100);

                unit16_t bgColor = ILI9341_BLACK;
                tft.fillRec(10,100,250,16,bgColor);

                tft.print("MODE : " + NRF_MODE);

LittleHackid, 2025-05-02:
> Sure ; that is what I would expect (i had the same issue on projects)
>
> After I make the change, how I recompile the code and upload?
>
> Also add here as well to clear the CH#
>
> ```
>    while [...]

Go to the Bruce-main folder and run "pio run" in the VSC terminal.

[image]

Try to comment out the boards you don't want to compile to save time and compilation errors.

[image]

kosmostarter, 2025-05-03:
I changed the platformio.ini and select just the CYD-2432s028 board.
I ran build in VSC and it completed and uploaded it to the display, but all I see is faint flashes.

I tried    "pio run"   in the terminal window and got.

PS C:\Users\xx\Downloads\Bruce-main> pio run
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is 
correct and try again.
At line:1 char:1
+ pio run
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
 
PS C:\Users\xx\Downloads\Bruce-main> 

Im getting closer , as soon as I figure out how to compile the code ,  it will open  up a lot , Would like to create a T-display enbed board.ini

kosmostarter, 2025-05-04:
Yes , I have that installed.

kosmostarter, 2025-05-04:
Im thinking on not selecting  the env.
This is the part Im not seeing.
"
In the Status bar (down on the screen) click where is written Default (Bruce-main)
A little Menu will appear in the Search bar (up on screen), select your device "env"
"

[image]

LittleHackid, 2025-05-04:
> Im thinking on not selecting the env. This is the part Im not seeing. " In the Status bar (down on the screen) click where is written Default (Bruce-main) A little Menu will appear in the Search bar (up on screen), select your device "env" "
> 
> [image]

You're doing well, my friend. Now you just have to go to the VSC terminal and navigate to the path where you have the platform.ini. Now in that terminal you can run pio run, but you need to be in the same path as bruce-main for it to work.
This is the extension I use and it works.
https://marketplace.visualstudio.com/items/?itemName=platformio.platformio-ide 
I was getting a long path error, I had to move the Bruce folder to the root of my hard drive, this solved it and I was able to use Pio