BruceDevices/firmware #1146: Universal Remote

CloudWaddie, 2025-04-29:
Bruce does not have this feature that the flipper zero has: universal remote.
I'd like a new app to be added named "Universal Remote". This will be able to load the .ir file from the flipper zero in universal remote format.

Alternatively, you can make copies of the file and "spam all" for one specific type by removing the other types. This is no where near as user friendly and is effectively another function.

https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/UniversalRemotes.md for reference.

CloudWaddie, 2025-04-30:
I can dev some of this if needed....

andreock, 2025-05-09:
That seems a nice feature, I will try to implement it soon

tommymakz420, 2025-06-28:
Cathacks managed to

Gosheto1234, 2025-07-11:
> > Bruce has a lot of features, and it is something that you can do with a JavaScript application to run on bruce..
> > I am against adding these universal codes into the flash memory, if you can run almost any .ir file from Flipper repository.. and flash size is something we must take care to keep it running on all devices and some of them are on the limit!
> > [image]
> 
> I understand this, however, couldn't you add the capability and provide instructions on where to get/make the files for it? This would allow the same functionality, whilst keeping size in mind.
you can get files from either from using reader module or you can download them from github , for example from here https://github.com/samallx/Hacking-Pack-/tree/main/M5StickHack-Pack/IR-Files , or from flipper zero IR DB , the files should work fine because they use the same formatting i think

CloudWaddie, 2025-07-24:
I understand that, but I am still wondering about the universal remote. Will this ever come?