BruceDevices/firmware #1150: What display is compatible with the ESP32-S3

kosmostrater, 2025-05-01:
I have an ESP32-S3 I like to use, but not sure which display is supported.
The ILI9341 or the TS7789?

And the connections.

I see in the ini.
static const uint8_t SS = 10;
static const uint8_t MOSI = 11;
static const uint8_t MISO = 13;
static const uint8_t SCK = 12;

But Where would DC and RST go for the display?
And the CS for Touchscreen?

I flashed the S3 and it shows up in the log windows, so now all I need is to connect the display

Thank you

[image]

ben7147, 2025-05-01:
Why would you need linux for that?

kosmostrater, 2025-05-01:
I read this in the readme.md and it looks like a Linux command

esptool.py --port /dev/ttyACM0 write_flash 0x00000 Bruce-<device>.bin

ben7147, 2025-05-01:
Thats not a linux command if it would be it would be this: ./esptool.py --port /dev/ttyACM0 write_flash 0x00000 Bruce-.bin

You can download eyltool for windows too btw and run this command from vs code's terminal... also use platformio for esp32

kosmostrater, 2025-05-02:
OK, thank you.  I'm Tring to learn how to compile the code  so I appreciate your help.

closed 2025-05-24 by bmorcelli

arfatkhilji513-eng, 2026-01-12:
I have esp s3 and 1.44 inch 128x128 RGB TFT st7735s this display worked with Bruce latest I flash latest Bruce not beta latest what I do next