BruceDevices/firmware #1153: ESP-32 generic not working with TFT ILI9341 SPI

Pand-R-Us, 2025-05-02:
Hi I'm pretty new to this stuff. I've been trying to connect my screen my ESP32 s3 and I've gone through the source code to try and match my wiring to the ESP32-generic defaults but i simply cant seem to get it to work. 

This is my pin layout.
[image]
I'm not quite sure what went wrong. Any help is greatly appreciated :)

LittleHackid, 2025-05-02:
You can try checking the library's pinout. Let's use the TFT_eSPI library as an example. By going to the library path and accessing the User_Setup.h file, you'll find that certain parameters are also defined here for your component to work, such as the driver and its pinout.
Check that the library you're using is made for your board. If not, configure it or search for your board's configuration file online.

[image]

[image]
I even think the pinout of your ESP32 and mine is different.
I'll help you with anything.

khaleeek, 2025-05-06:
Could you please share a code or guide how to connect esp32 s3 with ili or st77 display

Pand-R-Us, 2025-05-06:
Isn’t that the default driver? Like @LittleHackid said

> You can try checking the library's pinout. Let's use the TFT_eSPI library as an example. By going to the library path and accessing the User_Setup.h file, you'll find that certain parameters are [...]

closed 2025-05-26 by bmorcelli

khaleeek, 2025-05-26:
> not an issue, move to discussions

Bro please help me in code. Of esp 32 s3 with st7789 touch display 2.8 display