BruceDevices/firmware #1156: T-Embed CC1101 slows down *dramatically* after entering RF menu

SoLongSidekick, 2025-05-03:
**Describe the bug:**
After entering the RF menu and selecting any of the options the device freezes, then when it comes back it is so ungodly slow after moving the rotary encoder it might be 15-30 seconds before the menu actually responds. Only rebooting the device seems to fix the issue.

This behavior was visible in 1.9 and is also showing in the newest beta release.

Trofek321, 2025-05-13:
1.8.2 has this bug too

bmorcelli, 2025-05-13:
We are actually going to 1.10.. 

Test with Beta and tell us if it is working fine.. don't let to report after it gets released..

bmorcelli, 2025-05-14:
Fixed on Beta

closed 2025-05-14 by bmorcelli

Trofek321, 2025-05-14:
Thanks bro!

bmorcelli, 2025-05-14:
Send a video.. I can't reproduce it.. it is working for me in all devices

Huzzla101, 2025-05-18:
Its NOT Bruce fault... You have been into the config meny and changed most
likely Tx or Tx pin from 44 to 43 or other way or changed the CC1101
module. that will make the whole Lilygo go crash lag modus after you enter
RF meny, not before because it will not activate the pinouts before you
enter RF meny (DONT HELP RESTART BESIDES IT TAKE AWAY THE LAG UNTIL YOU
ENTER RF MENY AGAIN) . My advice is to RE INSTALL /FLASH bruce again and
when you enter the RF meny it should say at top of screen  CC1101  MODULE
and work fine again but  not mess with config unless needed....

Trofek321, 2025-05-18:
thanks bro!