BruceDevices/firmware #1162: bruce 1.9.1 core2 cc1101 nrf24 not showing in menu

on4ain, 2025-05-05:
I have a core2.1 item on which I want to run bruce.
I load bruce using https://bruce.computer/flasher (in chrome) or the m5burner app, but in both cases the cc1101 and nrf24 modules do not show up in the menus where I should select them.
It doesn't matter if I connect them or not, I cannot select them in the menu. 
How to proceed ?

on4ain, 2025-05-06:
some more testing: when I load the beta release via https://bruce.computer/flasher I do see nrf24 and cc1101 devices in menu, and when I select them I get this on the serial console:

[FROM M5Burner] /dev/ttyACM0 opened.
Selected: NRF24
[414077][V][common.cpp:1403] writeBytes(): [LGFX] i2c write error : ack wait
Selected: Main Menu
[455061][V][common.cpp:1449] readBytes(): [LGFX] i2c read error : ack wait
[455091][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Selected: RF
Selected: Config
[468746][V][common.cpp:1449] readBytes(): [LGFX] i2c read error : ack wait
[468826][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Selected: RF Module
Selected: CC1101
[477373][W][common.cpp:1480] readBytes(): [LGFX] i2c read error : read timeout

But when I load the latest release, nrf24 and cc1101 are not present in their respective menu's, and I get this in the console:

[FROM M5Burner] /dev/ttyACM0 opened.
[ 34819][V][common.cpp:1403] writeBytes(): [LGFX] i2c write error : ack wait
Selected: Config
Selected: RF Module
Selected: M5 RF433T/R
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 1,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
  "soundEnabled": 1,
  "wifiAtStartup": 0,
  "ledBright": 75,
  "ledColor": "0",
  "webUI": {
    "user": "admin",
    "pwd": "bruce"
  },
  "wifiAp": {
    "ssid": "BruceNet",
    "pwd": "brucenet"
  },
  "evilWifiNames": [],
  "bleName": "Keyboard_10",
  "wifi": {},
  "irTx": 32,
  "irTxRepeats": 0,
  "irRx": 33,
  "rfTx": 32,
  "rfRx": 33,
  "rfModule": 0,
  "rfFreq": 433.92,
  "rfFxdFreq": 1,
  "rfScanRange": 3,
  "rfidModule": 0,
  "mifareKeys": [],
  "gpsBaudrate": 9600,
  "startupApp": "",
  "wigleBasicToken": "",
  "devMode": 0,
  "colorInverted": 1,
  "disabledMenus": [],
  "qrCodes": [
    {
      "menuName": "Bruce AP",
      "content": "WIFI:T:WPA;S:BruceNet;P:brucenet;;"
    },
    {
      "menuName": "Bruce Wiki",
      "content": "https://github.com/pr3y/Bruce/wiki"
    },
    {
      "menuName": "Bruce Site",
      "content": "https://bruce.computer"
    },
    {
      "menuName": "Rickroll",
      "content": "https://youtu.be/dQw4w9WgXcQ"
    }
  ],
  "CC1101_Pins": {
    "sck": -1,
    "miso": -1,
    "mosi": -1,
    "cs": -1,
    "io0": -1,
    "io2": -1
  },
  "NRF24_Pins": {
    "sck": -1,
    "miso": -1,
    "mosi": -1,
    "cs": -1,
    "io0": -1
  },
  "SDCard_Pins": {
    "sck": 18,
    "miso": 38,
    "mosi": 23,
    "cs": 4,
    "io0": -1
  }
}[ 67388][I][config.cpp:266] saveFile(): config file written successfully
[ 67401][W][LittleFS.cpp:77] begin(): LittleFS Already Mounted!

Brown

bmorcelli, 2025-05-06:
They can be used on Core2 (I did it myself) but you need to configure the pins you will use to connect the CC1101 and NRF24 in the bruce.conf file.

```
"CC1101_Pins": {
"sck": 18,
"miso": 38,
"mosi": 23,
"cs": 32,
"io0": 33,
"io2": -1
},
"NRF24_Pins": {
"sck": 18,
"miso": 38,
"mosi": 23,
"cs": 32,
"io0": 33
},
"SDCard_Pins": {
"sck": 18,
"miso": 38,
"mosi": 23,
"cs": 4,
"io0": -1
}
```

This is an example using the SD Card sniffer and connecting to the portA grove (these pins are available on the pin header at the bottom too, so no need to use the sniffer)

on4ain, 2025-05-06:
tnx fer comment ! I reinstalled bruce latest release, and now I have the pins in bruce.conf as described by you.
when I select cc1101 without it connected,  there's no message on screen saying 'not found'. 
when I do connect cc1101 module, no message appears, but it's not working:
I believe I have wired everything correctly, but apparently some16:37:24.105 -> Selected: CC1101
16:37:30.294 -> Selected: RF
16:37:37.310 -> Selected: Spectrum
16:37:38.370 -> [199036][E][ELECHOUSE_CC1101_SRC_DRV.cpp:191] checkMISO(): CC1101: Fail checking MISO pin, check your connections
16:37:38.370 -> Rf Spectrum setup completething is wrong because I see this in serial console:

I have connected: 
cc1101 to co(re2:
pin 1 (gnd) to gnd (upper left)
pin 2 (vcc) to pin 3.3v (right column 6th pin from above)
pin 3 (GD0) to pin G33 (pa_scl)
pin 4 (CSN) to pin G32 (pa_sda)
pin 5 (SCK) to pin G18 (sck)
pin 6 (MOSI) to pin G23 (MOSI)
pin 7 (MISO) to pin G38 (MISO)

writing this I come to think that I probably have to swap G23 and G38. Is this correct ?

on4ain, 2025-05-06:
I have swapped G232 and G38, but still the same message in serial console, and spectrum does not work (is there an <esc> function somewhere to get out of the function ?

message in serial console:
7:09:44.570 -> Selected: RF
17:09:52.479 -> Selected: Spectrum
17:09:53.505 -> [ 23483][E][ELECHOUSE_CC1101_SRC_DRV.cpp:191] checkMISO(): CC1101: Fail checking MISO pin, check your connections

but the nrf24 does seem to work when connected to the same socket.

closed 2025-05-24 by bmorcelli