BruceDevices/firmware #1164: T-Embed front Panel.

kosmostrater, 2025-05-05:
I saw there was now a T-Embed on the webflash. I gave it a try and it loaded fine.

The issue is its not mounting the SDCARD.  log shows

[361404][W][sd_diskio.cpp:174] sdCommand(): no token received
[361510][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[361516][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[361522][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work

  "CC1101_Pins": {
    "sck": 38,
    "miso": 40,
    "mosi": 41,
    "cs": 43,
    "io0": 44,
    "io2": -1
  },
  "NRF24_Pins": {
    "sck": 38,
    "miso": 40,
    "mosi": 41,
    "cs": 43,
    "io0": 44
  },
  "SDCard_Pins": {
    "sck": 38,
    "miso": 40,
    "mosi": 41,
    "cs": 39,
    "io0": -1

SCK should be  40
MISO should be 38
According to the factor example from there website that does mount the SDcard

T-Embed isn't  in the boardlist yet in Bruce-Main or I would change it and recompile and see if it fixed it.

Thx
Kosmo

bmorcelli, 2025-05-05:
I fixed it in the Beta Version, can you test it?

kosmostrater, 2025-05-05:
I should have tried  that first , my bad.  yes it works. 
Next step is to connect my NRF24 and CC1101 to it :)

kosmostrater, 2025-05-05:
In the boards >  lilygo-t-embed-cc1101
  pins_ardunio.h

Line 150 #else // pins_arduino.h for T-Embed NON CC1101

Where do you  set the variable   to compile the NON CC1101  it as apposed to  the T-Embed C1101 board?

What to see if I can get the RGB working.
// RGB LED
#define HAS_RGB_LED 1
#define RGB_LED 14
#define LED_TYPE WS2812B
#define LED_ORDER GRB
#define LED_TYPE_IS_RGBW 0
#define LED_COUNT 8

I think "#define RGB_LED 14" should be  #define RGB_LED 42

Kosmo

kosmostrater, 2025-05-06:
Im still looking on how to compile for the non-cc1101 version. but in the pins in _ardunio.h under NON CC1101  look correct

// RGB LED
#define HAS_RGB_LED
#define RGB_LED 42
#define RGB_LED_CLK 45
#define LED_TYPE APA102
#define LED_ORDER RGB
#define LED_TYPE_IS_RGBW 0
#define LED_COUNT 8

But yea nothing shows on the LED's like   when you load the factory test SW from Lilygo

But there isnt really a menu that controls them either , that Ive found.

igniskarya, 2025-05-06:
the difference betwwn lilygo T embed CC1101 using ws2812 RGB....it will use the FastLED library, meanwhile for T embed non CC101 using APA102 RGB LED , refer to T embed, it has to use APA102.h (library)...is it right??

bmorcelli, 2025-05-07:
FastLed can handle this APA102.. and was set to work though.. maybe there's something wrong

https://github.com/pr3y/Bruce/blob/e1fe8c7d6e95ab08989f2bb64aafd01254dfb231/boards/lilygo-t-embed-cc1101/pins_arduino.h#L280

igniskarya, 2025-05-09:
@bmorcelli ..on pin arduino.h for APA102 LED_ORDER RGB, meanwhile from some references :

"_Define the color order (usually BGR for APA102): #define COLOR_ORDER BGR._ "

bmorcelli, 2025-05-09:
I don't have this device to test, so of you could build it locally and find out if these fixes work.. feel free to show us the solution and open a PR with the fixes 😉

igniskarya, 2025-05-09:
alright then.. I'll try

bmorcelli, 2025-05-26:
duplicated https://github.com/pr3y/Bruce/issues/1132

closed 2025-05-26 by bmorcelli