BruceDevices/firmware #1167: Spelling mistake

EagerToLearnBrit, 2025-05-07:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

Minor misspelling

andreock, 2025-05-07:
Which version are you using? I'm on latest firmware(compiled from source) and I can't see any mistakes under WiFi > EvilPortal

andreock, 2025-05-07:
> EagerToLearnBrit left a comment [(pr3y/Bruce#1167)](https://github.com/pr3y/Bruce/issues/1167#issuecomment-2859317611)
>
> [Screenshot_20250507_175321_DuckDuckGo.jpg (view on web)](https://github.com/user-attachments/assets/354a3d41-b342-4bfb-befe-82dbc89edffc)

OK, you mean on wiki. I will open a PR later to fix it.

-------- Messaggio originale --------

EagerToLearnBrit, 2025-05-07:
Yes mate

closed 2025-05-13 by EagerToLearnBrit