BruceDevices/firmware #1168: ir_brute.js is not working

Gosheto1234, 2025-05-07:
**Describe the bug:**
i tried to run ir_brute.js on m5stickcplus2 but the IR was not working , nothing was blinking even on camera , the screen was only changing

**Steps To Reproduce:**

- Device used: M5stickcplus2

**Expected behavior:**
it should use the IR transmitter and send signals 

**Actual behavior:**
it doesnt use the IR transmitter 

**Additional context / logs:**

eadmaster, 2025-05-09:
it was broken in this refactoring: https://github.com/pr3y/Bruce/commit/d13c6fed308d6887895a433d0db2f9c9d5616f39

need to change line 24 like this:

`serialCmd("ir send {\"Protocol\":\"" + protocol + "\",\"Bits\":32,\"Data\":\"0x" + curr_val + "\"}");`

eadmaster, 2025-05-16:
update: still broken, needs another fix to correctly parse the json

closed 2025-05-18 by bmorcelli

Gosheto1234, 2025-05-18:
after the update , when i try to run ir_brute.js it just doesnt start it just exits the .js , tested on the latest beta with m5stickcplus2

eadmaster, 2025-05-19:
> after the update , when i try to run ir_brute.js it just doesnt start it just exits the .js , tested on the latest beta with m5stickcplus2

this seems an unrelated regression, also affecting other scripts...
it should be related to some of the changes in this commit:
https://github.com/eadmaster/Bruce/commit/d8eb107fe4cdba729602b92ac61a54fbd4fd4093

PR #1208 (2025-05-18): serial cmds refactorings and fixes
- fixed #1168 
- restored Tasmota-like `IRSend` and `RFSend` serial cmds 
- refactoring: use ArduinoJson instead of cJSON
- fixed  `storage write` serial cmd
- added `irTransmit()` and `subghzTransmit()` JS functions 
- cleanups