BruceDevices/firmware #1170: bruce FM name

tiramisu42, 2025-05-08:
Hello, I noticed that when intercepting a frequency in some machines, the frequency name changes to Bruce FM. Can you please add a function or add a config tab where you can change this name to your own?

Krutifulmen, 2025-05-08:
Yo, i see you at tiktok :0

tiramisu42, 2025-05-09:
> Yo, i see you at tiktok :0

:)

tiramisu42, 2025-05-09:
I would also like to listen to the radio using a speaker on the same t-embed

bmorcelli, 2025-05-09:
> I would also like to listen to the radio using a speaker on the same t-embed

It is not possible with this module, it is Transmitter only, unfortunately 

What is your idea of a customized name? Can it be set each session? So you just set the name and start the radio?.. because I don't see the necessity to keep it persistent (saved in bruce.conf)

Other thing I've been discussing, is to move this whole FM section to a JavaScript, and then give the freedom to deal with JS file easily

tiramisu42, 2025-05-09:
but how does spectrum work then?

so you want to add a name to bruce.conf?

bmorcelli, 2025-05-09:
> but how does spectrum work then?

It reads the Noise level in each frequency and draw this value into the screen

It can't digitalize the audio and stream it to through I2C to the MCU.. the same way we cannot stream data from the MCU (such as mp3, wav) directly through I2C to the module.. we need to insert the audio data through the P2 connector and it will add into the carrier frequency

> so you want to add a name to bruce.conf?

No, I don't.. I would put it as a variable in the context, just it.. it would reset after reboot