BruceDevices/firmware #1176: esp32-s3 rebooting error

khaleeqmalik786786, 2025-05-09:
after flashing in esp32-s3 n16r8 
getting this error

after flashing generic esp32 firmware  getting this error

after flashing bruce geeting this in serial console
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c5a
  #0  0x40378c5a in Cache_Suspend_ICache at /home/runner/esp32-arduino-lib-builder/esp-idf/components/esp_rom/patches/esp_rom_cache.c:56

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x118c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc20
load:0x403cb700,len:0x30e0
entry 0x403c88b8
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420029a3  PS      : 0x00060230  A0      : 0x82002a7c  A1      : 0x3fca07b0  
A2      : 0x3fc962b0  A3      : 0x00000000  A4      : 0x0000000c  A5      : 0x0000000d  
A6      : 0x000000ff  A7      : 0x60004000  A8      : 0x00000010  A9      : 0x08000000  
A10     : 0x3fc963c4  A11     : 0x019bfcc0  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x019bfcc0  A15     : 0x3fca0788  SAR     : 0x00000002  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x420029a0:0x3fca07b0 0x42002a79:0x3fca07e0 0x42001b0d:0x3fca0800 0x42006fda:0x3fca0830 0x4037b986:0x3fca0850
  #0  0x420029a0 in _stext at ??:?
  #"1"  0x3fca07b0 in _lx_intr_stack at esp_ipc_isr_handler.S.obj:?
  #"2"  0x42002a79 in _stext at ??:?
  #3  0x3fca07e0 in _lx_intr_stack at esp_ipc_isr_handler.S.obj:?
  #4  0x42001b0d in _stext at ??:?
  #5  0x3fca0800 in _lx_intr_stack at esp_ipc_isr_handler.S.obj:?
  #6  0x42006fda in ArduinoJson::V741PB22::detail::JsonSerializer<ArduinoJson::V741PB22::detail::Writer<HWCDC, void> >::visit(decltype(nullptr)) at .pio/libdeps/esp32-s3-devkitc-1-psram/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:102
      (inlined by) ArduinoJson::V741PB22::detail::PrettyJsonSerializer<ArduinoJson::V741PB22::detail::Writer<HWCDC, void> >::result_type ArduinoJson::V741PB22::detail::VariantData::accept<ArduinoJson::V741PB22::detail::PrettyJsonSerializer<ArduinoJson::V741PB22::detail::Writer<HWCDC, void> > >(ArduinoJson::V741PB22::detail::PrettyJsonSerializer<ArduinoJson::V741PB22::detail::Writer<HWCDC, void> >&, ArduinoJson::V741PB22::detail::ResourceManager const*) const at .pio/libdeps/esp32-s3-devkitc-1-psram/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:108
  #7  0x3fca0830 in _lx_intr_stack at esp_ipc_isr_handler.S.obj:?
  #8  0x4037b986 in r_lld_adv_ext_chain_scannable_construct at ??:?
  #9  0x3fca0850 in _lx_intr_stack at esp_ipc_isr_handler.S.obj:?

ELF file SHA256: b44d65fea

Rebooting...

bmorcelli, 2025-06-19:
Fixed on Beta (18/06/25)

khaleeek, 2025-06-19:
How to get webui on esp32 s3 n16r8

khaleeek, 2025-06-20:
Thank you if i can connect st7789 2.8in tft or ili9341 2.8in tft display so it is more handy to use it.

khaleeek, 2025-06-20:
after flashing i use serial terminal and use following command 
wifi add Pixel 12345678
where Pixel is SSID 
an 12345678 is password
but still unable to launch webui

Steuriger, 2025-06-23:
Yeah im also Trying to make one with ili9341 2,8in Display but didnt Work but now i will try the headless Version looking forward to IT!

eadmaster, 2025-06-24:
Try the command to start the webui:
`webui`

khaleeek, 2025-06-24:
[image]

Nothing happens

eadmaster, 2025-06-24:
check your serial connection speed and parameters: 
speed = 115200
8 data bits, 1 stop bit, no parity, no Flow Control

eadmaster, 2025-06-24:
other cmds like `ls` works?

khaleeek, 2025-06-24:
[image]

[image]

Still same