BruceDevices/firmware #1181: Friends and "Find Bruce devices Nerbay"

Krutifulmen, 2025-05-10:
If your friend have a device with bruce frimware, you can add it to friends lost.

Find Bruce Devices nerbay:
You can search device with a bruce frimware nerbay you

You or other people can send friend request. If you get a request you can accept it or deny. Also, you can send it.

If you added someone as a friend, you can see what device friend had. If friend nerbay: what is he doing (ble spam, wifi attack or other). And, of course unfriend him or even block.

I hope you like my function and it will be added.
Sorry for my bad english

tommyfart, 2025-05-15:
and you can also message friends and with a messaging app and friends app in Connect section

eadmaster, 2025-05-15:
related: https://github.com/pr3y/Bruce/issues/974

Could be implemented with BLE or Wifi.