BruceDevices/firmware #1183: add t-embed cc1101 more orientation

wr102102, 2025-05-10:
Can you add more orientation for the screen? For example, as shown in the picture, I think the vertical version is more suitable for one-handed operation. Now I can only achieve it by modifying the configuration file, and the screen content overlaps

[image]

bmorcelli, 2025-05-12:
Interesting..

Tell us how things get into the screen, like RF, RFID and IR captured signals, Menus and etc..

if possible send us pictures

The limitation of orientation in made here, it will allow portrait mode if WIDTH and HEIGHT are greater than 240 pixels: 
https://github.com/pr3y/Bruce/blob/7b2c7894eaf28592a745994d641fa20b6847bb21/src/core/settings.cpp#L52C2-L52C42

bmorcelli, 2025-05-15:
I added it now, let it available for this device.. some screens might be buggy, maybe

closed 2025-05-15 by bmorcelli