BruceDevices/firmware #1185: 1.10.0 Beta Release issues.

bmorcelli, 2025-05-11:
Report bugs from Beta Release here.

fail4k, 2025-05-11:
**Describe the bug:**
unable to run js script

**Steps To Reproduce:**
try to run any js script in the js interpreter menu
- Device used: Lilygo T-Embed CC1101

**Expected behavior:**
run js script

**Actual behavior:**
when I try to launch it, it just crashes to the main menu

**Additional context / logs:**
in order for the script to be able to run, you need to go into the files and select a specific .zhs file and run it from the files

fail4k, 2025-05-11:
**Describe the bug:**
signal recording problem

**Steps To Reproduce:**
go to the RF section and then to the raw record
- Device used: Lilygo T-Embed CC1101

**Expected behavior:**
it was expected that after recording the signal I would be able to: play, reject, save it

**Actual behavior:**
If the signal is recorded for a long time, then after pressing the "ok" button the device reboots. If the signal is short, then a menu appears with a choice of what to do with the signal, but the choice of saving and replay leads to a reboot without the ability to save

**Additional context / logs:**
there are isolated cases when everything works out and it can save the signal or play it, but only if the signal is short

th1s-david, 2025-05-11:
https://github.com/pr3y/Bruce/issues/1184#issue-3055108663

fail4k, 2025-05-11:
**Describe the bug:**
oddities with ble spam

**Steps To Reproduce:**
choose any type of spam
- Device used: Lilygo T-Embed CC1101

**Expected behavior:**
just spam, no extra problems

**Actual behavior:**
BLE spam and no effect on the ok button

**Additional context / logs:**
if you hold down the ok button when attacking, then for some reason something happens when nothing should happen

djaysan, 2025-05-11:
Is the latest beta, the one you can flash from the online flasher?
if so, there is an issue with GPS on t-display s3 (non touch).

with latest stable, gps works but on beta it does not anymore - i tested with 2 devices and the same gps unit. it just times out and says no gps.

djaysan, 2025-05-11:
Describe the bug:
Run NRF24 - issue with additional button

Steps To Reproduce:
try to run NRF24 > NRF Jammer or CH Jammer with additional button, just goes back to main screen. but if i stay press on confirm button then it works

Device used: Lilygo T-Display S3 (non touch)
Expected behavior:
Open NRF Jammer or CH Jammer
Actual behavior:
It just goes back to main screen. but if i stay press on confirm button then it works

Additional context / logs:
with latest stable, it works with additional button. confirmed and tested.

tegtheevi, 2025-05-11:
-Bruce gotchi crashes in beta 
-When sd card is plugged and you capture a rf signal in autosave it keeps capturing fake signals (sometimes)
-bruce airgeddon keeps crashing when target inputes credentials

bmorcelli, 2025-05-12:
> Describe the bug:
> GPS not working
> 
> Steps To Reproduce:
> try to run wardriving or GPS Tracker
> 
> Device used: Lilygo T-Display S3 (non touch)
> Expected behavior:
> get sat. & getting location
> 
> Actual behavior:
> when I launch wardriving or GPS Tracker - it does not get location, says no gps. can't quite wardriving either with any button
> 
> Additional context / logs:
> with latest stable, gps works but on beta it does not anymore - i tested with 2 devices and the same gps unit. it just times out and says no gps. 

I changed the pinouts for this board, and I intend to remove the button on gpio 16 to give room for all modules..

Actually theres a conflict with gpio 16 with the button and Serial TX (used for gps), as I moved it from 44/43 to 16/17

https://github.com/pr3y/Bruce/blob/4a8a497348abf01a7e49184664f7c7456299fcd0/boards/lilygo-t-display-s3/pins_arduino.h#L117

007n7-roblox, 2025-05-12:
Describe the bug:
Run CC1101 - Machine freeze

Steps To Reproduce:
try to run CC1101 RF>conf>RF module CC1101,it says cc1101 not found And hangs

Device used: M5 stick c plus 2
Expected behavior:
Open CC1101 RF
Actual behavior:
CC1101 not found And hangs

Additional context / logs:
Can fix cc1101 error and work stably And also no hang up confirmed and tested.

silentfox2023, 2025-05-15:
Describe the bug:
Scan/Copy Function problem

Steps To Reproduce:
go to the RF section and then to the Scan/Copy

Device used: Bruce Smoochiee V2
Expected behavior:
it was expected that after recording the signal I would be able to: play, reject, save it

Actual behavior:
The device reboots after pressing the Next key.

Additional context / logs:
In Bruce Smoochiee V1, it works normally and can display the function menu.

silentfox2023, 2025-05-15:
Describe the bug:
There is lag when selecting various function menus problem

Steps To Reproduce:
Lags occur during the selection of WiFi, Ble, NRF24, etc.

Device used: Bruce Smoochiee V2
Expected behavior:
No delay, no need for multiple key presses, and no lag.

Actual behavior:
The menu can only be moved by pressing the keys more than twice, and this applies to all keys.

Additional context / logs:
In Bruce Smoochiee V1, it works normally.

silentfox2023, 2025-05-17:
Describe the bug
no charging status icon, and it cannot be charged, no charging indicator light.

Steps To Reproduce:
During the whole process of use

Device used: Bruce Smoochiee V2 and Bruce Smoochiee V1

Expected behavior:
Normal charging and displaying battery status.

Actual behavior:
The replaced versions are 1.9 and 1.8.2, and the charger has been replaced.

Additional context / logs:
The same problem also exists in Bruce Smoochiee V1. Purchase components completely according to the BOM list, weld them according to the schematic diagram, and there are no situations such as false soldering or missing soldering in the welding process.

luckyy218, 2025-07-27:
Bruce Beta on m5 stick c plus 2. 
When you have a theme on m5 stick and go to files, then Web ui. When you click exit or run in background, then it restarts all the time, so I can't Navigatie from mobile phone.

peppemas1, 2025-08-27:
with last beta MASS STORAGE dont work , device T-embed cc1101, i select mass storage, then connect cable , when cable usb connected t-embed reset. 
Device T-embed cc1101
windows 11
if i flash a previous beta Mass Storage work perfect

Toster-cpu, 2025-09-22:
Firmware 1.1 don't work in m5stickc plus 1.1

salarinichimpanzini, 2026-11-03:
hello the website to dowload the firmware https://bruce.computer/flasher dosent work with waveshare i want to dowload the firmware and connect but donr work whene i want to dowload the firmware to flash without the web flasher i says  page not found