BruceDevices/firmware #1187: NRF24 on T-Embed CC1101 does not work

Mawa000, 2025-05-12:
I have installed an NRF24 module on my T-Embed, and although the WiFi and the rest of the functions are working fine, the NRF is not working.
The installation of the module seems correct because the firmware detects the board and allows me to open the spectrum analyzer and the Jammer. The problem is that I do not detect any signal, and the Jammer does not seem to work on any frequency.

I leave you a link to the development of the installation in case you can identify any problems with the physical installation, as it seems to me to be a software issue.
https://github.com/Xinyuan-LilyGO/T-Embed-CC1101/issues/30

bmorcelli, 2025-05-12:
Have you tried Beta version?

peppemas1, 2025-05-12:
By default, the module nrf24 gt-24 that you install uses the integrated PCB antenna low range only fee cm. The chip inductor is soldered to 1, 2 as shown in the figure. If you need to use the ipx holder to extract the external antenna, you can put the chip inductor to 1,3 as shown.

[image]

danformiga, 2025-05-12:
> Have you tried Beta version?

Estou tendo um problema parecido, fiz um conector de 8 pinos para alternar entre o NRF24 e o CC1101, o CC1101 funciona perfeitamente, mas quando coloco o NFR24, ele detecta, acessa as funções, mas não aparecem as barras do Spectrum e nada funciona. Já viu algo parecido? Estou usando um CYD B1.9.1.

I'm having a similar issue. I made an 8-pin connector to switch between the NRF24 and the CC1101. The CC1101 works perfectly, but when I connect the NRF24, the functions are accessible but Spectrum bars don't appear and nothing works. Have you seen anything like this? I'm using a CYD B1.9.1.

Mawa000, 2025-05-13:
> De forma predeterminada, el módulo nrf24 gt-24 que instala utiliza la antena PCB integrada de bajo alcance solo unos pocos cm. El inductor de chip está soldado a 1, 2 como se muestra en la figura. Si necesita usar el soporte ipx para extraer la antena externa, puede poner el inductor del chip a 1,3 como se muestra.
> 
> [image]

Indeed, I forgot to change the bridge for the IPX antenna.... X( I was aware of it, but I skipped that step during the process. I will modify it and share the changes. I'm impressed by the very low emission level of the PCB antenna, they could just do without it XD.
Thank you very much for your help, buddy.

Mawa000, 2025-05-13:
> ¿Has probado la versión Beta?

Yes, I think I am working with the latest available Beta version. Apparently, the problem comes from me not changing the bridge that selects the external antenna on the board, as indicated by @peppemas1

peppemas1, 2025-05-13:
yes the pcb antenna on gt-24 as very low range only few centimetres 20-30 , i replace the gt-24 with the GT24 mini with only ipex connector on board, i get 3-5 meters jammer with 3dbi antenna

Mawa000, 2025-05-13:
> yes the pcb antenna on gt-24 as very low range only few centimetres 20-30 , i replace the gt-24 with the Gt-24 smd mini with only ipex connector on board, i get 3-5 meters jammer with 3dbi antenna

It's not a lot either, but according to the information I've seen, it is to be expected.
I suppose that to cover greater transmission distances a lot of power would be necessary, which is not compatible with the consumption we require in these devices.

bmorcelli, 2025-05-13:
So, if changing the antenna solved your issue, close this Issue please..

Mawa000, 2025-05-14:
I finally changed the switch to use the IPX connector instead of the PCB antenna and it has improved the range. However, the coverage is still limited, but I suppose it is due to a hardware limitation.
I consider the issue resolved. In the future, I will try to install a higher power one like the E01-2G4M27D. I will keep you updated on the progress.

closed 2025-05-14 by Mawa000