BruceDevices/firmware #1189: wifi signal on beta bruce

peppemas1, 2025-05-13:
I noticed that in the latest beta versions, the wifi signal reception has decreased, when I scanned in the same spot I detected many more networks now only 2, I use t-embed cc1101 with external wifi antenna. What has changed in the new beta releases?

peppemas1, 2025-05-13:
very strange , I noticed this problem with the latest beta, tonight I'll try to install 1.9.1, thx

peppemas1, 2025-05-14:
an other user on discord channel (bruce - wifi) have noticed same issue , please Pirata more check on this

eadmaster, 2025-05-14:
try to reflash the old version and repeat the test

peppemas1, 2025-05-14:
i have try 1.9.1 and same problem, doing also factory reset, always low wifi signal

eadmaster, 2025-05-14:
so it is not dependent on the code, it's something in your test setup.

peppemas1, 2025-05-14:
maybe an hardware problem, i have tested other external antenna and the inside antenna, official 1.9.1 and older beta version

peppemas1, 2025-05-14:
i close the issue , seems only a my device problem, i have try many firmware version also othje like capibara, and always low wifi signal

peppemas1, 2025-05-15:
i have try all possible things, firmware capibara and stock, original antenna, i noticed if i enable evil portal work at long range more of 40mt , but in receiver mode wifi have low strenght

HalfMatt, 2025-05-15:
Can confirm I'm having a similar problem, using beta  `dfc51d7`
Tested on three different 2.4ghz external antennae(Router, alfa, and nrf24l01 antenna) 
It seems like the receiving part is the only issue, but I can broadcast fine
Currently at -93db on my home AP when it used to be at -20db with the same setup.

peppemas1, 2025-05-15:
try now to remove fat , reflash launcher and install fresh last beta, nothing

peppemas1, 2025-05-16:
i have read esp32 pdf an found this:
Core Reset: resets the whole digital system except RTC, including CPU0, CPU1, peripherals, Wi-Fi,
Bluetooth® LE (BLE), and digital GPIOs.
question, how is possible to do this core reset on esp32?

eadmaster, 2025-05-20:
> i want also try to send serial cmd to increase wifi tx power

There is no such cmd currently, and i'm not sure if it would really help:

> the ESP32 will default to maximum WiFi TX power after a reset or power on. 

(source)[https://esp32.com/viewtopic.php?t=35835]

If it is something related to the Bruce code, we need more info to reproduce the issue (e.g. a firmware version where the range is better)

Possible similar issue: https://github.com/espressif/arduino-esp32/discussions/11088

peppemas1, 2025-05-21:
wifi is bidirectional but i see in the manual esp32 is possible to setting rx and tx parameters

[image]

some value is possible to set to more gain in RX and TX

peppemas1, 2025-05-21:
ai say this about tx power default on esp32 s3 wroom1 , default is setting only to 9dbm

[image]

peppemas1, 2025-05-22:
my tests report better wifi performance, this serial cmds will be implemented in the official beta ?

closed 2025-05-24 by bmorcelli