BruceDevices/firmware #1195: cyd ESP32-2432S028, but with 1.9.1 the colors

jhonpohac, 2025-05-15:
Hello friends, I need your help. I had 1.8.1 installed and it worked fine on my cyd ESP32-2432S028, but with 1.9.1 the colors are wrong. I've already installed all of them and it doesn't work properly. Please help me so I can have it right and enjoy it. We hope for more modifications and improvements. Thank you very much

[image]

[image]

[image]

[image]

you very much

bmorcelli, 2025-05-15:
Config> UI Color> Inverted Colors

jhonpohac, 2025-05-15:
> Configuración > Color de la interfaz de usuario > Colores invertidos

Hello friend, thank you very much, excellent, since it turned out really well for me. Excellent work, I hope they continue releasing updates, adding more and more things that we can all enjoy. Thank you very much, excellent work, it continues to be the best system.

bmorcelli, 2025-05-15:
Tyvm!

closed 2025-05-15 by bmorcelli