BruceDevices/firmware #1204: Lilygo T display S3  GPS

doyavivo, 2025-05-16:
I try flashing Lilygo T display S3  no SD MMC with beta version..GPS not working..click Menu FILES on SD Card, Go back to Main menu ..so I re-flashing to latest version.. GPS working SD card menu still Go back to the main menu..

any suggestion?

bmorcelli, 2025-05-16:
I changed the pinouts, will revert it this evening

djaysan, 2025-05-16:
Good to hear @bmorcelli - is the c1101 wired differently? i'm trying to hook the cc11001 module only - no sd or gps. is there a proper pin out out there? and what gpo should i set in RF>config for RX & TX?
Cheers! keep up the amazing work!

bmorcelli, 2025-06-10:
https://github.com/pr3y/Bruce/blob/main/boards/lilygo-t-display-s3/connections.md

closed 2025-06-11 by doyavivo