BruceDevices/firmware #1205: Graphics editor for u8g2, AdafruitGFX, Arduino, TFT_eSPI

akeendev, 2025-05-16:
https://lopaka.app/gallery/5684/12494

static const unsigned char PROGMEM image_battery_0_bits[] = {0x00,0x00,0x00,0x0f,0xff,0xfe,0x10,0x00,0x01,0x10,0x40,0x41,0x70,0x20,0x81,0x80,0x11,0x01,0x80,0x0a,0x01,0x80,0x04,0x01,0x80,0x0a,0x01,0x80,0x11,0x01,0x70,0x20,0x81,0x10,0x40,0x41,0x10,0x00,0x01,0x0f,0xff,0xfe,0x00,0x00,0x00,0x00,0x00,0x00};

static const unsigned char PROGMEM image_battery_17_bits[] = {0x00,0x00,0x00,0x0f,0xff,0xfe,0x10,0x00,0x01,0x10,0x00,0x0d,0x70,0x00,0x0d,0x80,0x00,0x0d,0x80,0x00,0x0d,0x80,0x00,0x0d,0x80,0x00,0x0d,0x80,0x00,0x0d,0x70,0x00,0x0d,0x10,0x00,0x0d,0x10,0x00,0x01,0x0f,0xff,0xfe,0x00,0x00,0x00,0x00,0x00,0x00};

static const unsigned char PROGMEM image_battery_33_bits[] = {0x00,0x00,0x00,0x0f,0xff,0xfe,0x10,0x00,0x01,0x10,0x00,0x6d,0x70,0x00,0x6d,0x80,0x00,0x6d,0x80,0x00,0x6d,0x80,0x00,0x6d,0x80,0x00,0x6d,0x80,0x00,0x6d,0x70,0x00,0x6d,0x10,0x00,0x6d,0x10,0x00,0x01,0x0f,0xff,0xfe,0x00,0x00,0x00,0x00,0x00,0x00};

static const unsigned char PROGMEM image_battery_50_bits[] = {0x00,0x00,0x00,0x0f,0xff,0xfe,0x10,0x00,0x01,0x10,0x03,0x6d,0x70,0x03,0x6d,0x80,0x03,0x6d,0x80,0x03,0x6d,0x80,0x03,0x6d,0x80,0x03,0x6d,0x80,0x03,0x6d,0x70,0x03,0x6d,0x10,0x03,0x6d,0x10,0x00,0x01,0x0f,0xff,0xfe,0x00,0x00,0x00,0x00,0x00,0x00};

static const unsigned char PROGMEM image_battery_67_bits[] = {0x00,0x00,0x00,0x0f,0xff,0xfe,0x10,0x00,0x01,0x10,0x1b,0x6d,0x70,0x1b,0x6d,0x80,0x1b,0x6d,0x80,0x1b,0x6d,0x80,0x1b,0x6d,0x80,0x1b,0x6d,0x80,0x1b,0x6d,0x70,0x1b,0x6d,0x10,0x1b,0x6d,0x10,0x00,0x01,0x0f,0xff,0xfe,0x00,0x00,0x00,0x00,0x00,0x00};

static const unsigned char PROGMEM image_battery_83_bits[] = {0x00,0x00,0x00,0x0f,0xff,0xfe,0x10,0x00,0x01,0x10,0xdb,0x6d,0x70,0xdb,0x6d,0x80,0xdb,0x6d,0x80,0xdb,0x6d,0x80,0xdb,0x6d,0x80,0xdb,0x6d,0x80,0xdb,0x6d,0x70,0xdb,0x6d,0x10,0xdb,0x6d,0x10,0x00,0x01,0x0f,0xff,0xfe,0x00,0x00,0x00,0x00,0x00,0x00};

static const unsigned char PROGMEM image_battery_charging_bits[] = {0x00,0x02,0x00,0x0f,0xe4,0xfe,0x10,0x0c,0x01,0x10,0x08,0x01,0x70,0x18,0x01,0x80,0x30,0x01,0x80,0x3f,0x81,0x80,0x7f,0x01,0x80,0x03,0x01,0x80,0x06,0x01,0x70,0x04,0x01,0x10,0x0c,0x01,0x10,0x08,0x01,0x0f,0xd3,0xfe,0x00,0x10,0x00,0x00,0x00,0x00};

static const unsigned char PROGMEM image_battery_full_bits[] = {0x00,0x00,0x00,0x0f,0xff,0xfe,0x10,0x00,0x01,0x16,0xdb,0x6d,0x76,0xdb,0x6d,0x86,0xdb,0x6d,0x86,0xdb,0x6d,0x86,0xdb,0x6d,0x86,0xdb,0x6d,0x86,0xdb,0x6d,0x76,0xdb,0x6d,0x16,0xdb,0x6d,0x10,0x00,0x01,0x0f,0xff,0xfe,0x00,0x00,0x00,0x00,0x00,0x00};

static const unsigned char PROGMEM image_bike_bits[] = {0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x1c,0x00,0x07,0x04,0x00,0x02,0x08,0x00,0x79,0xf8,0x00,0x3b,0x2e,0x00,0x45,0x59,0x00,0x83,0xa4,0x80,0x93,0x24,0x80,0x82,0xa0,0x80,0x44,0xd1,0x00,0x38,0x0e,0x00};

static const unsigned char PROGMEM image_bluetooth_connected_bits[] = {0x01,0x00,0x02,0x80,0x02,0x40,0x22,0x20,0x12,0x20,0x4a,0x48,0x26,0x90,0x33,0x30,0x26,0x90,0x4a,0x48,0x12,0x20,0x22,0x20,0x02,0x40,0x02,0x80,0x01,0x00,0x00,0x00};

static const unsigned char PROGMEM image_network_not_connected_bits[] = {0x82,0x0e,0x44,0x0a,0x28,0x0a,0x10,0x0a,0x28,0xea,0x44,0xaa,0x82,0xaa,0x00,0xaa,0x0e,0xaa,0x0a,0xaa,0x0a,0xaa,0x0a,0xaa,0xea,0xaa,0xaa,0xaa,0xee,0xee,0x00,0x00};

tft.setTextColor(0xFFFF);
tft.setTextSize(1);
tft.setFreeFont();
tft.drawString("U8g2 on RT-thread", 2, 7);

tft.drawBitmap(2, 20, image_battery_0_bits, 24, 16, 0xFFFF);

tft.drawBitmap(54, 20, image_battery_33_bits, 24, 16, 0xFFFF);

tft.drawBitmap(28, 20, image_battery_17_bits, 24, 16, 0xFFFF);

tft.drawBitmap(80, 20, image_battery_50_bits, 24, 16, 0xFFFF);

tft.drawBitmap(2, 41, image_battery_67_bits, 24, 16, 0xFFFF);

tft.drawBitmap(108, 39, image_bike_bits, 17, 16, 0xFFFF);

tft.drawBitmap(27, 41, image_battery_83_bits, 24, 16, 0xFFFF);

tft.drawBitmap(54, 41, image_battery_full_bits, 24, 16, 0xFFFF);

tft.drawBitmap(81, 41, image_battery_charging_bits, 24, 16, 0xFFFF);

tft.drawBitmap(109, 22, image_bluetooth_connected_bits, 14, 16, 0xFFFF);

tft.drawBitmap(108, 3, image_network_not_connected_bits, 15, 16, 0xFFFF);

bmorcelli, 2025-05-16:
??

Huzzla101, 2025-05-18:
> Hey, as the founder of Lopaka, I'm glad to see it's useful for Bruce. Feel
> free to ask any questions or suggest a feature.

Yes if you are the founder of Lopaka , then please feel free to Help Mr
Pirata and all of us make Bruce even Greater with contributing with small,
useful codes that is allways needed and small cool icons or helpful small
things that we need but didnt know we needed before we did know that we
.... you get the point.. =)