BruceDevices/firmware #1207: t-display-s3 + sd card(spi) + nrf24 = problem

alrek12345, 2025-05-16:
I use dev last version bruce.
I connect sd card slot (spi) + nrf24, if I insert sd card in slot, after reboot esp32-s3 - "NRF24 not found", but sd card working.
If I remove sd card and reboot esp32-s3, NRF24 working... and sd card reading too

- Device used: 
t-display-s3

My settings in pins_arduino.h:

#define SPI_SS_PIN 10
#define SPI_MOSI_PIN 11
#define SPI_MISO_PIN 13
#define SPI_SCK_PIN 12

#define SDCARD_CS SPI_SS_PIN
#define SDCARD_SCK SPI_SCK_PIN
#define SDCARD_MISO SPI_MISO_PIN
#define SDCARD_MOSI SPI_MOSI_PIN

#define USE_NRF24_VIA_SPI
#define NRF24_CE_PIN 3
#define NRF24_SS_PIN 2
#define NRF24_MOSI_PIN SPI_MOSI_PIN
#define NRF24_SCK_PIN SPI_SCK_PIN
#define NRF24_MISO_PIN SPI_MISO_PIN

How I can fix it?

Huzzla101, 2025-05-18:
Bro.. It CLEARLY says in description under legal use that you have to be
Minimum Level 3 to use Bruce.....So level up first..... JOKING.... But can
your device handle 2 attachments at once? Im not sure. Also SD card need to
read and send out data, how do you integrate this with NRF24 that also need
Master in and out, together its 14pinouts, you can use the same ground for
all or the last NRF24 pinout  is perfect for extra ground nr2... But you
will have a interference /comunication problem with data in and out for SD
AND master in out for NRF at same time for such small device

alrek12345, 2025-05-18:
Different SPI Pins Must Be Used for Devices: sd and nrf24

closed 2025-05-18 by alrek12345