BruceDevices/firmware #1211: bad ble not work  on latest bruce beta

y4zdev, 2025-05-18:
**Describe the bug:**
when enable bad bluetooth that device not show properly as "keyboard_1c" but some times its show as a sring
[image]

**Steps To Reproduce:**
1. flash bruce beta "4f995e8" fw to m5stick.
2. then go to bluetooth >> bad ble >> select text payload("any thing") > select **us** keyboard layout >> wait 
3.  scan bluetooth using other device (pc/mobile)
- Device used: m5stick plus 2

**Expected behavior:**
show Bluetooth device name "keyboard_c1" to connect

**Actual behavior:**
showing a sting instead of the "keyboard_1c" also its not able to connect

**Additional context / logs:**
i don't have any logs, srry

Adam-neeeds-help, 2025-05-26:
Still not working

bmorcelli, 2025-05-27:
Patched it now, can you test?

Adam-neeeds-help, 2025-05-27:
> OK I'll reflash and get back to you
> If you want to test, yes, BetaRelease

thx ive reflashed and bad ble works now