BruceDevices/firmware #1217: DIY Smoochiee board not capture RF signal

HiennNek, 2025-05-19:
**Describe the bug:**
My DIY Smoochiee board is not capturing any RF signal when I press my remote using the latest beta version (commit `af4c30c`). But the RF capture still works on version 1.9.1.

**Steps To Reproduce:**
Go to `RF`
Press `Scan/copy`
- Device used: DIY Smoochiee board using v2 schematic

**Expected behavior:**
Capture my remote signal with RCSwitch protocol

**Actual behavior:**
Nothing captured

**Additional context / logs:**
When I go to `Config` -> `RF Module` -> `CC1101`, it can still detect the RF module.
Also, the latest beta version `af4c30c` RF capture still work on CYD board.

ilbossdelfaidate, 2025-05-21:
if you don't have this component on board AW9523BTQR , rf will not work or reboot device random standard cc1101 have this harware issue

HiennNek, 2025-05-22:
@ilbossdelfaidate, thank you for your reply.
I had tried to comment out `USE_IO_EXPANDER` and `IO_EXPANDER_AW9523` in `boards`>`smoochiee-board`>`pins_arduino.h`. But it doesn't work. Also, I'm only using a CC1101 module and no NRF24 module, so there is no shared GPIO pin, and each component on my board has a separate GPIO. Here is what I have on my DIY board:
- 1.9-inch ST7789 TFT LCD screen
- 5 Button
- An IR LED
- IR receiver
- PN532 NFC module
- CC1101 module

closed 2025-05-26 by bmorcelli