BruceDevices/firmware #1218: JS Interpreter issues

AnoNy-Sec, 2025-05-19:
**Describe the bug:**
When I try to run JS scripts "ir_brute.js", "rf_brute.js", "wifi_brute.js" by @eadmaster from sd_files/interpreter/ library, instead of working properly it opens script's menu for a less than a second and then it crashes to the main menu. All three scripts run with the same problem, no error message or anything, it just crashes the script

**Steps To Reproduce:**
JS Interpreter > Load... > Select one of the scripts > Experience the bug.
- Device used: T-Embed CC1101

**Expected behavior:**
I expected all three JS scripts to run smoothly (some do)

**Actual behavior:**
After running them I see script's menu for a second and then it crashes me to main menu for some reason

eadmaster, 2025-05-19:
followup https://github.com/pr3y/Bruce/issues/1168#issuecomment-2889470336

It should be something affecting to the `loopOptions` function

bmorcelli, 2025-05-27:
Fixed the dialogChoice functionality by patching the loopOptions function.

Updated the rf_brute.js, ir_brute.js and wifi_brute.js

Available on Beta for testing

closed 2025-05-31 by bmorcelli