BruceDevices/firmware #1224: Implement Card Skimmer Detection

jk31454, 2025-05-21:
**Is your feature request related to a problem? Please describe.**

We go on roadtrips over the summer, and our card gets stolen 2-3 times from gas station card skimmers. ESP32marauder has a feature to scan for them, however I normally use Bruce for the larger featureset. It would be great for this to be implemented in Bruce.

**Describe the solution you'd like**

From this article: https://learn.sparkfun.com/tutorials/gas-pump-skimmers/all#characteristics-of-the-device

"Identified Commands:

    P - returns M
    ~ - Erase all SPI flash. This is how to erase all the credit card numbers. Unit blinks the status LED for ~20 seconds (EEPROM takes time to erase). The unit will buffer any incoming serial characters during the time it takes to erase the EEPROM (serial interrupts and buffer are being used)."

From https://github.com/justcallmekoko/ESP32Marauder/wiki/detect-card-skimmers

"Essentially you just need to scan for Bluetooth devices. If there is a password protected device with any of the names from the list above, there is a chance it is a credit card skimmer. Since the people who use these skimmers are skids at best, the firmware found on them is likely very cookie cutter and will have the same default password. If you can connect to the Bluetooth device and send a "P", you will know if you have a skimmer once you receive an "M"."

My proposed solution is implementing a program that would detect these skimmers, access them and then wipe their SPI flash.

Huzzla101, 2025-05-21:
Well its a cool feature but let me tell you something.... I know for sure
that all cards in the world are for sale..... Bankers and hackers  etc have
done this for years..Call it Illuminati if u like...... But since it has
gone from track2 on magnetic stripe to the chip  instead, that decreased
the fraud usage alot.... Just letting you know.... So always look for small
creditcard amounts charged that you cant remember you have done as they
either empty card at once or charge small amounts monthly to look legit....
Also you can close card for usage outside your region.....

jk31454, 2025-05-21:
I am aware, however we are unable to get new cards while we are camping, it's annoying to have to check, and it would be better to prevent this before it happens rather than after, even if it only makes up a small portion of total stolen cards.

Huzzla101, 2025-05-21:
True indeed ...