BruceDevices/firmware #1225: Record raw issues

DevEclipse1, 2025-05-21:
**Describe the issue**
If i record a remote control, even a wireless doorbell or such, and i save it and replay it from the file, it doesnt do stuff as the remote control (it doesnt have rolling codes)

**Steps To Reproduce:**
Record any remote that doesnt have rolling codes and replay it from a saved file
- Device used: 
Diy board with cc1101

**Expected behavior:**
Send the code and replicate the remote/doorbell

**Actual behavior**
It sends something but doesnt do stuff as the remote cntrl

**Additional context / logs:**

Huzzla101, 2025-05-21:
Since your doorbell is a brand and most likely well known.... Have you
tried to record the signal as NOT raw to , as if the subGHZ have any
records in NOT raw section they may recognize your doorbell there to then
the signal that will be sendt is slightly different from the raw
capture...But raw capture should work to.... Have you checked that you
sending the raw signal at correct frequency etc? As i think that even when
you capture a RAW signal it will only duplicate it BUT  send it out at your
selected frequency that may not fit with your doorbell...  Im just trying
find solution here my dude .....

Huzzla101, 2025-05-21:
Because you say that it send out "something" and that may be the recorded
signal just at wrong frequency, check if it is same freq you send out as
the doorbell using

closed 2025-05-26 by bmorcelli