BruceDevices/firmware #1227: Wifi and bluetooth not work and cause reboot on smoochie board in version c86b22c

ilbossdelfaidate, 2025-05-21:
Wifi and bluetooth not work and cause reboot on smoochie board in version c86b22c
hot spot not work since first beta after 1.9.1
i beacon reboot device

RF/IR/NRF24/GPS/RFID/SD/SCREEN/BUTTON work great

HiennNek, 2025-05-23:
Hi, after I did some research, if you had a clone ESP32-S3 board with a weird-looking antenna, you would have some issues with the processing performance of the board and lower memory size than on the specs. 

[image]

Try to buy a genuine ESP32-S3 board from Espressif Systems, or buy a board that has the Espressif logo on it from VCC-GND studio, or try this purple board that worked completely fine for me

[image]

Also, a bad solder joint or shorting component on the board, especially with the 5 navigation buttons, could make the ESP32-S3 work very slowly (idk why), and it does happen to me when I build my DIY board. So you should check it first by connecting everything to the breadboard before soldering it.

ilbossdelfaidate, 2025-05-23:
bug solved on f22640e

only Wifi-Ap  and eviltwin don't work
beacon spam , bluetooth, mass storage, webui etc... work fine in this version

closed 2025-05-23 by ilbossdelfaidate