BruceDevices/firmware #1234: Does the lilygo t-embeded cc1101 support bad usb trough USB A?

VladGamer43, 2025-05-23:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

Huzzla101, 2025-05-23:
Yes you have the bluetooth options AND in OTHERS  (in meny )  you have the
normal bad usb with usb wire options to .. =)

VladGamer43, 2025-05-23:
But does the lilygo support the wired one?

Huzzla101, 2025-05-23:
yes in your meny go to  OTHERS/BADUSB  that is the wired usb method where u
must have a usb wired connection... But it also have the possibility to do
the same with connected bluetooth....

VladGamer43, 2025-05-23:
I know there is an option but how to connect it, and I DONT want blotooth

Huzzla101, 2025-05-23:
simply put a usb wire between ur lilygo and the device then push badusb and
then you can choose from your stored base of bad usb and use any of them

closed 2025-05-24 by bmorcelli