BruceDevices/firmware #1248: Error while displaying wifi network information

LetsEatLead, 2025-05-25:
When you go to wifi, wifi attacks, target attacks, select a wifi network, information: The screen appears to display the information for a fraction of a second, then it reverts back to the screen where you select the attack. If you click information quickly, the screen turns black besides a empty battery icon and crashes, then it restarts.

Information:
Cardputer v1.1 
M5Launcher
Bruce 1.10 (Not Beta)

Serial Log:
18:40:18.259 -> Selected: WiFi
18:40:21.163 -> Selected: Wifi Atks
18:40:22.036 -> Selected: Target Atks
18:40:28.664 -> [ 61735][V][WiFiGeneric.cpp:384] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 19
18:40:28.664 -> [ 61744][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 1 - SCAN_DONE
18:40:30.670 -> Selected: #WifiNetwork (-46|WPA2/PSK)
18:40:31.962 -> Selected: Information
18:40:33.121 -> Selected: Information
18:40:33.544 -> Selected: Information
18:40:34.061 -> Selected: Information
18:40:34.254 -> Selected: Information
18:40:34.479 -> Selected: Information
18:40:34.931 -> Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
18:40:34.931 -> Debug exception reason: Stack canary watchpoint triggered (loopTask) 
18:40:34.931 -> Core  1 register dump:
18:40:34.931 -> PC      : 0x4038614f  PS      : 0x00060536  A0      : 0x803851fb  A1      : 0x3fcea3a0  
18:40:34.931 -> A2      : 0x3fc9f1d0  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060523  
18:40:34.931 -> A6      : 0x00060523  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff  
18:40:34.931 -> A10     : 0x4214870c  A11     : 0x00000020  A12     : 0x000000ab  A13     : 0x0000000c  
18:40:34.931 -> A14     : 0x02c9f1d0  A15     : 0x00ffffff  SAR     : 0x0000001f  EXCCAUSE: 0x00000001  
18:40:34.931 -> EXCVADDR: 0x00000000  LBEG    : 0x4214884c  LEND    : 0x4214885d  LCOUNT  : 0x00000000  
18:40:34.931 -> 
18:40:34.931 -> 
18:40:34.931 -> Backtrace: 0x4038614c:0x3fcea3a0 0x403851f8:0x3fcea3e0 0x40382d96:0x3fcea400 0x4038323a:0x3fcea420 0x42154e19:0x3fcea460 0x420bc59f:0x3fcea480 0x42148951:0x3fcea4a0 0x422987f9:0x3fcea4f0 0x4214cb31:0x3fcea520 0x4214cca5:0x3fcea570 0x4214ccf5:0x3fcea590 0x42010b65:0x3fcea5d0 0x42010cb9:0x3fcea630 0x4201126b:0x3fcea6d0 0x4208ed45:0x3fcea6f0 0x4208ef1f:0x3fcea740 0x4201010e:0x3fcea790 0x42011606:0x3fcea7b0 0x4201f22f:0x3fcea820 0x4208df87:0x3fcea840 0x4208ee7e:0x3fceaa80 0x4208ef1f:0x3fceaad0 0x4201010e:0x3fceab20 0x42011606:0x3fceab40 0x4201f22f:0x3fceabb0 0x4208df87:0x3fceabd0 0x4208ee7e:0x3fceae10 0x4208ef1f:0x3fceae60 0x4201010e:0x3fceaeb0 0x42011606:0x3fceaed0 0x4201f22f:0x3fceaf40 0x4208df87:0x3fceaf60 0x4208ee7e:0x3fceb1a0 0x4208ef1f:0x3fceb1f0 0x4201010e:0x3fceb240 0x42011606:0x3fceb260 0x4201f22f:0x3fceb2d0 0x4208df87:0x3fceb2f0 0x4208ee7e:0x3fceb530 0x4208ef1f:0x3fceb580 0x4201010e:0x3fceb5d0 0x42011606:0x3fceb5f0 0x4201f22f:0x3fceb660 0x4208df87:0x3fceb680 0x4208ee7e:0x3fceb8c0 0x4208ef1f:0x3fceb910 0x4201010e:0x3fceb960 0x42011606:0x3fceb980 0x4201f22f:0x3fceb9f0 0x4208df87:0x3fceba10 0x4208e059:0x3fcebc50 0x4201010e:0x3fcebcb0 0x42011606:0x3fcebcd0 0x4201f22f:0x3fcebd40 0x4208e56d:0x3fcebd60 0x4228ff31:0x3fcebf10 0x4201010e:0x3fcebf30 0x42011606:0x3fcebf50 0x4201cbd1:0x3fcebfc0 0x4228feed:0x3fcec0a0 0x4201010e:0x3fcec0c0 0x42011606:0x3fcec0e0 0x4201401e:0x3fcec150 0x42039b62:0x3fcec1e0 0x42157ba8:0x3fcec220
18:40:34.931 -> 
18:40:34.931 -> 
18:40:34.931 -> 
18:40:34.931 -> 
18:40:34.931 -> ELF file SHA256: f0546d590467a3f9
18:40:34.931 -> 
18:40:34.931 -> Rebooting...

nickvicious, 2025-05-27:
Have same thing on 1.10 on T-Embed CC1101

bmorcelli, 2025-05-27:
Fixed now on Beta