BruceDevices/firmware #1304: My battery capped

TP566, 2025-05-31:
**Describe the bug:**
My battery capped at 74%

**Steps To Reproduce:**

- Device used:T embed CC1101 

**Expected behavior:**
Supposed to be 100%

**Actual behavior:**
Capped at 74%

**Additional context / logs:**

bmorcelli, 2025-05-31:
Have you tried a better charger?

Charging on USB computer is not enough for me after the last update, because now we have fixed the capacity reader and restablished the factory chip charger setup.

closed 2025-05-31 by bmorcelli

bmorcelli, 2025-06-06:
Appearently the BQ27220 is loosing itself, and you can "fix it" doing some tricks:
* Remove the battery, and reconnect it 2 minutes later, to make sure BQ27220 resets
* Make your battery dies, keeping it on until it turns off.. it will make BQ27220 to reset
* (maybe, didnt test this option) Use the Turn Off option, and turn on again.. but i don't thing it would help, but worth trying