BruceDevices/firmware #1318: SD Card Bug on Lilygo T-embed CC1101

notveryeffect, 2025-05-31:
**Describe the bug:**

When I turn on the device, the SD Card is there but when i try to use it, Files>SD Card, the main menu pops up.

**Steps To Reproduce:**

- Device used: Lilygo T-embed CC1101
- SD Card: Lenovo G32
- Firmware: 1.10.1

**Expected behavior:**

Found every file in the SD Card.

**Actual behavior:**

I go back on the main menu and the voice "SD Card" is completely gone.

**Additional context / logs:**

Sometime the SD Card runs but a bit slowly.
I've already tried to format it (FAT32) but its the same.

bmorcelli, 2025-05-31:
It probably is a problem with your SD Card... 

You must use a SDHC lesser than 32Gb formatted in FAT32.. 

Sdxc and high speed cards are not supported (or will lead to issues) with any ESP32, nit only bruce.

notveryeffect, 2025-05-31:
Its a thing of the last FW ? Because I've used it on my lilygo for a long time.

Huzzla101, 2025-05-31:
Then reflash FW without SD card inside then turn it off and insert card
(32gb at most and SDHC will work 99% of the time even the unbranded ones
yes SDHC not normal SD) and when its off enter SD card and turn on... wait
a little as card need to feel welcome and if after 3min no SD card logo
appear then go to files and check if can enter SD card and if not then open
the back of t embed and push the RST button once......Lets hope that will
find ur sd card, if not then turn it off and on again)YES sometimes lilygo
need some time to get comfortably with a card becoue it works with bruce
LIVE)....  Another trick is to enter the sd card while using Bruce at same
time but thats last resort as that may also resort to some kind of SD card
bug...

Huzzla101, 2025-05-31:
Ahh you  used it on ur lilygo a long time? then its just to flash from beta
to normal or other way and it should be ok..

closed 2025-06-19 by bmorcelli