BruceDevices/firmware #1321: Ble spam non working on lilygo t-deck plus

pitro64, 2025-06-01:
**Describe the bug:**
Every ble spam mode (ios, windows, samsung, android and all) doesn't work at all

**Steps To Reproduce:**
(Normal use) turn on the t-deck go under ble and select a spam mode

**Expected behavior:**
Spamming ble devices of the selected mode;

**Actual behavior:**
It says thats spamming but it does nothing

**Additional context / logs:**
Bruce version 1.10.1 and 9043efa (I don't know if it works with other versions because I just got the t-deck)

Media cmds,ble scan,i beacon,bad ble,ble keyboard, these modes works

Device used: Lilygo T-deck plus with LORA 868MHz

With M5StickC Plus2 everything works!

pitro64, 2025-06-01:
Why wontfix ?

bmorcelli, 2025-06-01:
Because it is what it is.. it is not what you are expecting... iOS, Android and Microsoft have already patched it and it is not efficient anymore..

I am very prone to exclude it from Bruce completely.. 

I've beed answered this very same question many times, and you don't really try to understand what is going on..

So wait for the next version without this feature..

pitro64, 2025-06-01:
"In a few minutes I'll send you a video.

bmorcelli, 2025-06-01:
Before start spamming, restart your mobile phone, and keep t-deck close to you mobile

bmorcelli, 2025-06-01:
If your mobile detects this kind of threat, it stops showing the advertisement.. thats why I'm telling you to restart your phone and try with the T-Deck.

Same happens with other devices.. the code running is the same to all, so it is not device related, it is the target protecting themselves.

pitro64, 2025-06-01:
"I realized that it doesn't change from the device I use because it's a problem that happens on all devices since version 1.10.0."

pitro64, 2025-06-01:
How do i send you the video?

pitro64, 2025-06-01:
Do not remove the spam function because it still works.
Only in version 1.10.1 and 1.10.0 ble spam doesn't work