BruceDevices/firmware #1322: Megalodon bugs on t-deck

pitro64, 2025-06-01:
**Describe the bug:**
1.With lilygo t-deck plus when i play megalodon and with the shark i go over the touch button "prev" it eats a pice of it

2.after 100 points the touch commands for megalodon stop working (after the game says so...)

3.the trakball has inverted input with in megalodon and main menu 
Megalodon(scroll down=shark goes up scroll up=shark goes down)
Main menu(scroll left=next left page scroll right=next right page)

**Steps To Reproduce:**
1.Every time i play megalodon
2.every time i get to the secret tools menu in the game
3.normal trakball use

**Expected behavior:**
1. The text should remain in place
2. The touch shoudl continue working
3. Megalodon(scroll down=shark goes down scroll up=shark goes up)
Main menu(scroll left=next right page scroll right=next left page)

**Actual behavior:**
1.The text disappear
2.the touch doesn't work anymore
3.Megalodon(scroll down=shark goes up scroll up=shark goes down)
Main menu(scroll left=next left page scroll right=next right page)

**Additional context / logs:**
Bruce version 1.10.1
Device lilygo t-deck plus

bmorcelli, 2025-06-04:
> 1.With lilygo t-deck plus when i play megalodon and with the shark i go over the touch button "prev" it eats a pice of it

Didn´t undestand it

> 2.after 100 points the touch commands for megalodon stop working (after the game says so...)

it is waiting for "select" to be pressed.. so you must tap the center of the  screen, of press the trackball, or press Enter on the keyboard to go on...

> 3.the trakball has inverted input with in megalodon and main menu Megalodon(scroll down=shark goes up scroll up=shark goes down) Main menu(scroll left=next left page scroll right=next right page)

yes.. it is indeed.. i'm working on it.

pitro64, 2025-06-04:
1. I would send you a photo or video but I do not know how

2.I mean, when you start playing again after all the writing, sometimes the touch doesn't work.

closed 2025-06-11 by bmorcelli