BruceDevices/firmware #1330: Wifi functions R broken on CYD

tegtheevi, 2025-06-02:
**Describe the bug:**
 wifi section is not working at all 

**Steps To Reproduce:**

- Device used: cyd 2usb

**Expected behavior:**

**Actual behavior:**

**Additional context / logs:**

tegtheevi, 2025-06-06:
@bmorcelli idk why my wifi section always crashes can someone check please

bmorcelli, 2025-06-06:
I have no Idea... I have a standard CYD (not 2Usb) and it doesn't happen..

Maybe your WiFi interface has some kind of problem...

Did Launcher OTA or other firmware worked good with it?

tegtheevi, 2025-06-06:
@bmorcelli im not sure why  it happens evert time i go to wifi  it crashes and reboots 
Ima try to figure it out and not waste your time 
If i figure out the problem ill point it out

closed 2025-06-15 by tegtheevi