BruceDevices/firmware #1334: Rf scan not working

Shivam00817, 2025-06-02:
When I connect my Rd modules cc1101 to m5 stick c plus2 and configure it as "cc1101 <shared spi> it shows error that c1101 not round even when wiring is correct, i changed jumper wires few times swapped module to see if another c1101 may work but the result stays same no matter what i did . also sometimes it doesn't show any error when I click on configure>shared spi> without connecting module it doesn't detect properly please resolve this !!

tegtheevi, 2025-06-02:
Have you tried to change the module in bruce 
In the rf section go to config>rf module and change the module from m5 433 to cc1101

Shivam00817, 2025-06-03:
Hello ! Thankyou for your response it's working in legacy mode perfectly