BruceDevices/firmware #1358: Green battery icon

pitro64, 2025-06-05:
**Describe the bug:**
Randomly,mabye is with time,the border of the battery icon turns lime green and after a while it goes back to the ui color selected

**Steps To Reproduce:**
It's random 

**Expected behavior:**
Shoudln't happen

**Actual behavior:**

**Additional context / logs:**
Happends from bruce version 1.10.0
Device lilygo t-deck plus

bmorcelli, 2025-06-06:
T-deck doesn't have a fuelGauge or charger chip to detect if it is charging or not, so I made a workaround to identify it reading the bus voltage each 30 s, if the voltage is higher than before, probably it is charging..

I will remove it, because it is not working as expected.

bmorcelli, 2025-06-06:
Fixed on Beta