BruceDevices/firmware #1369: m5stickcplus2 double click on m5button

Gosheto1234, 2025-06-07:
**Describe the bug:**
after i connected to my wifi when i tried going into wifi options , it double clicked and it disconnected me from my wifi , it happened 3 times in a row , the device is m5stickcplus2 , also after i try to connect and it doesnt connect (says wifi offline) after i retry and it connects the menu speed when i hold down next button is super fast 

https://github.com/user-attachments/assets/ec31faf7-9c18-47b8-a570-c12c699514e6

bmorcelli, 2025-06-07:
It kept LongPress variable set.. 

Need more information about how it started to replicate and debug

Gosheto1234, 2025-06-07:
it started right after i connected to my wifi network which is hidden , after it connected me , when i went back to the menu it started speeding up and double pressing , if i press next button it like skips 2 times (like i pressed the button 2 times) , that is what i know for this bug at the moment , if i find anything else i will text ASAP

Gosheto1234, 2025-06-08:
i found something that maybe can lead to answear the glitch happens after you type something in the keyboard and then you click okay , like after you type something in the keyboard and send it , this bug can be canceled if you press previous button once

bmorcelli, 2025-06-09:
I found a few places where LongPress variable wasn't being rearmed.. maybe it is fixed now..

beta is building

Gosheto1234, 2025-06-10:
Its fixed , thanks

closed 2025-06-10 by Gosheto1234