BruceDevices/firmware #1373: Mic Rec error

vlqdus, 2025-06-09:
**Describe the bug:**
Mic recording feature error "Failed to create directory" on the microSD card, even though there is already a directory
**Steps To Reproduce:**
Others -> Mic Recording (must have microSD card plugged in) -> error "Failed to create directory"
- Device used: 
M5StickC P2
**Expected behavior:**
Should record sounds to a file in /microSD/BruceMIC/file001.wav 
**Actual behavior:**
Doesn't even create nor detect the folder on the microSD card

vlqdus, 2025-06-09:
its a 1gb microSD, laptop detects it perfectly, I'll try another one and let you know

Huzzla101, 2025-06-09:
Then  try re flash to beta version, or if u already using beta then flash
it back to default version and try again. If it dont make a map for it
becouse its allready there then most likely that will happen 1 out of
10000000 times so you may have won the Brude lottery ... =)

Huzzla101, 2025-06-09:
Bruce*lottery --- And also if that dont work then reflash again but DELETE
the whole map that is already there .....
Maybe that is all that is needed to .... So Brice can be confident when he
want to build a new folder

Huzzla101, 2025-06-09:
BRUCE*