BruceDevices/firmware #1375: wifi attack

deiniel10, 2025-06-09:
When i use wifi attack and then use deauth+clone on my other router where i login my device,after that when i try to connect again after the deauth the captive portal of evil twin not show but the wifi deauth work same on deauth+clone+verify same captive portal not show but if i forget password on my wifi and try to login with deauth+clone+verify on it show hope you understand sorry for my bad english

bmorcelli, 2025-06-09:
sorry, didn't understand the problem

write in your own language into google translator and paste here (if it is spanish or portuguese, paste here in these languages as well)

naofumi39538, 2025-06-11:
**Here's the clarification words of what she's saying:**

When I perform a Wi-Fi attack and then run deauthentication plus cloning on my other router (the one where my device is logged in), the captive portal for the evil twin does not appear when I try to reconnect after the deauth. The deauthentication still works, but even if I use deauth + clone + verify, the captive portal does not show up. However, if I first “forget” the Wi-Fi network on my device and then attempt to log in using deauth + clone + verify, the captive portal appears. Hope this makes sense—sorry for my bad English.