BruceDevices/firmware #1382: Lilygo T-Embed CC1101 not starting

LevinBoss18, 2025-06-10:
The Lilygo T-Embed CC1101 that i have will turn off if it is not on a power source. If it is it will refuse to turn off. If you turn it off, it will simply not start again. It will just start if you connect it via a cable. So there is no way you can turn on your lilygo, if you wont leave it on standy all the time. That was not an issue in previous versions, please fix.

bmorcelli, 2025-06-11:
you probably is using it the wrong..

if you keep it plugged to USB and use the `Config> Turn Off` option.. it will turn off the power ship, but won't turn off the device (it is still powered by usb, but once you remove from usb it will shut down and will turn on only when you plug it on the usb cord again..

if you use the `Config> Deepsleep` (or keep the upper button pressed long enaugh) it will "shut down" immediately.. entering in deep sleep.. and will wake up when you press the upper button

but it is only possible if your battery is charged and plugged accordingly.. I just tested everything here, and everything is working fine

LevinBoss18, 2025-06-14:
The Same Issue is when I Try to get it into Deep sleep. Turns Off Never turns back on